fangism/hackt #40: gcc 5.2.1 doesn't like the STL wrappers

cornell-cs5620, 2015-11-21:
In file included from ./util/sublist.hh:11:0,
                 from <command-line>:0:
./util/STL/list_fwd.hh:22:10: error: reference to ‘list’ is ambiguous
  typedef list<T, std::allocator<T> > type;
          ^
./util/STL/list_fwd.hh:18:7: note: candidates are: template<class T, class Alloc> class std::list
 class list;
       ^
In file included from /usr/include/c++/5/list:63:0,
                 from ./util/sublist.hh:10,
                 from <command-line>:0:
/usr/include/c++/5/bits/stl_list.h:507:11: note:                 template<class _Tp, class _Alloc> class std::__cxx11::list
     class list : protected _List_base<_Tp, _Alloc>
           ^
Makefile:7090: recipe for target 'util/sublist.hhchk' failed

nbingham1, 2015-11-21:
Eh, I was logged into the wrong account... Oh well.

nbingham1, 2015-11-24:
It seems as though this applies for clang as well (This was a fresh clone):

./bootstrap
./configure CXX=clang CC=clang
make

./util/STL/list_fwd.hh:22:10: error: unknown type name 'list'
        typedef list<T, std::allocator<T> >     type;
                ^
./util/STL/list_fwd.hh:22:14: error: expected member name or ';' after declaration specifiers
        typedef list<T, std::allocator<T> >     type;
        ~~~~~~~~~~~~^

nbingham@Lira:~/src/hackt$ clang --version
Ubuntu clang version 3.6.2-1 (tags/RELEASE_362/final) (based on LLVM 3.6.2)
Target: x86_64-pc-linux-gnu
Thread model: posix

fangism, 2016-05-03:
I actually looked at this a few months ago, and started trying to make my forward declaration headers work with GNU libstdc++ and (clang) libc++, and it was a significant (unfinished) effort.  I'm thinking that maybe it's not worth maintaining these forward declaration headers -- one option is to scrap them in favor of including standard full headers (<list>, <map>) at the expense of increased compile time in some cases.

cornell-cs5620, 2016-05-06:
> scrap them in favor of including standard full headers (, ) at the expense

Increased compile time for something shouldn't often be compiled is not
such a terrible thing.

tadeoman, 2017-05-08:
I just ran into this same error while compiling with gcc 6.3.1. Is there a workaround?

fangism, 2017-12-15:
Somehow this message escaped my attention.  Workaround is to replace these silly headers with the original header, like <list>.  Need to find time to modernize the codebase...