fangism/hackt #52: Segmentation fault

nbingham1, 2016-06-30:
#FILE "x.hac" %{
  pint TOOL_SIM = 1;
  pint TOOL_NETLIST = 2;
%}
#FILE "y.prs" %{
#FILE "z.prs" %{
#FILE "w.prs" %{
      preal pn = 2.0;
      pint lsz = 10;
%}

defproc inv(bool? in; bool! out) {
}

export template <pint N><pint sz> defproc nand(bool? in[N]; bool! out) {
  prs {
    (|:i:N: ~in[i]<pn*sz> ) -> out+
  }
}

export template <pint N><pint sz> defproc nor(bool? in[N]; bool! out) {
  prs {
    (|:i:N:  in[i]<sz>      ) -> out-
    (&:i:N: ~in[i]<N*pn*sz> ) -> out+
  }
}

export template <pint N; pbool invert><pint sz> defproc andN(bool? in[N]; bool! out);
export template <pint N; pbool invert><pint sz> defproc orN(bool? in[N]; bool! out);
export template <pint N; pbool invert><pint sz> defproc andN(bool? in[N]; bool! out) {
  [ N == 1 ->
    (;i:N/2-N%2:
      sub2[i]<sz>(in[i*2..i*2+1], mid[i]);
    )
  ]
}

defproc orN(bool? in[N]; bool! out) {
  [ N == 1 ->
  ]
}

defproc asymmetric_celem_cfb(bool? dn[ND], up[NU]; bool! out) {
}

export defchan clocked_bit <: chan(bool) (bool clk, d) {
}

%}
%}

nbingham1, 2016-06-30:
#0  0x00007ffff411a7e3 in ?? () from /lib/x86_64-linux-gnu/libc.so.6
#1  0x00007ffff411bce8 in ?? () from /lib/x86_64-linux-gnu/libc.so.6
#2  0x00007ffff411e6c0 in malloc () from /lib/x86_64-linux-gnu/libc.so.6
#3  0x00007ffff49dbdad in operator new(unsigned long) () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007ffff6c83cd4 in __gnu_cxx::new_allocator<std::_List_node<util::memory::fixed_pool_chunk<HAC::entity::pint_const, 32ul> > >::allocate(unsigned long, void const*) () from /auto/edatools/hackt/20160622/lib/hackt/libhacktparsers.so.0
#5  0x00007ffff6c83c19 in std::_List_base<util::memory::fixed_pool_chunk<HAC::entity::pint_const, 32ul>, std::allocator<util::memory::fixed_pool_chunk<HAC::entity::pint_const, 32ul> > >::_M_get_node() () from /auto/edatools/hackt/20160622/lib/hackt/libhacktparsers.so.0
#6  0x00007ffff6c83b10 in std::list<util::memory::fixed_pool_chunk<HAC::entity::pint_const, 32ul>, std::allocator<util::memory::fixed_pool_chunk<HAC::entity::pint_const, 32ul> > >::_M_create_node(util::memory::fixed_pool_chunk<HAC::entity::pint_const, 32ul> const&) () from /auto/edatools/hackt/20160622/lib/hackt/libhacktparsers.so.0
#7  0x00007ffff6c83ad1 in std::list<util::memory::fixed_pool_chunk<HAC::entity::pint_const, 32ul>, std::allocator<util::memory::fixed_pool_chunk<HAC::entity::pint_const, 32ul> > >::_M_insert(std::_List_iterator<util::memory::fixed_pool_chunk<HAC::entity::pint_const, 32ul> >, util::memory::fixed_pool_chunk<HAC::entity::pint_const, 32ul> const&) () from /auto/edatools/hackt/20160622/lib/hackt/libhacktparsers.so.0
#8  0x00007ffff6c6dc95 in std::list<util::memory::fixed_pool_chunk<HAC::entity::pint_const, 32ul>, std::allocator<util::memory::fixed_pool_chunk<HAC::entity::pint_const, 32ul> > >::push_back(util::memory::fixed_pool_chunk<HAC::entity::pint_const, 32ul> const&) () from /auto/edatools/hackt/20160622/lib/hackt/libhacktparsers.so.0
#9  0x00007ffff6c6da61 in util::memory::chunk_map_pool<HAC::entity::pint_const, 32ul, true>::allocate() () from /auto/edatools/hackt/20160622/lib/hackt/libhacktparsers.so.0
#10 0x00007ffff6c6014c in HAC::entity::pint_const::operator new(unsigned long) () from /auto/edatools/hackt/20160622/lib/hackt/libhacktparsers.so.0
#11 0x00007ffff6aaf212 in HAC::parser::token_int::check_meta_expr(HAC::parser::context const&) const () from /auto/edatools/hackt/20160622/lib/hackt/libhacktparsers.so.0
#12 0x00007ffff6aaf2f6 in non-virtual thunk to HAC::parser::token_int::check_meta_expr(HAC::parser::context const&) const () from /auto/edatools/hackt/20160622/lib/hackt/libhacktparsers.so.0
#13 0x00007ffff6a63a18 in HAC::parser::arith_expr::check_meta_expr(HAC::parser::context const&) const () from /auto/edatools/hackt/20160622/lib/hackt/libhacktparsers.so.0
#14 0x00007ffff6a63a18 in HAC::parser::arith_expr::check_meta_expr(HAC::parser::context const&) const () from /auto/edatools/hackt/20160622/lib/hackt/libhacktparsers.so.0
#15 0x00007ffff6a99951 in HAC::parser::range::check_meta_index(HAC::parser::context const&) const () from /auto/edatools/hackt/20160622/lib/hackt/libhacktparsers.so.0
#16 0x00007ffff6afd698 in HAC::parser::loop_instantiation::check_build(HAC::parser::context&) const () from /auto/edatools/hackt/20160622/lib/hackt/libhacktparsers.so.0
#17 0x00007ffff6b07f41 in HAC::parser::node_list<HAC::parser::instance_management const>::check_build(HAC::parser::context&) const () from /auto/edatools/hackt/20160622/lib/hackt/libhacktparsers.so.0
#18 0x00007ffff6afec26 in HAC::parser::guarded_instance_management::check_build(HAC::parser::context&) const () from /auto/edatools/hackt/20160622/lib/hackt/libhacktparsers.so.0
#19 0x00007ffff6b09291 in HAC::parser::node_list<HAC::parser::guarded_instance_management const>::check_build(HAC::parser::context&) const () from /auto/edatools/hackt/20160622/lib/hackt/libhacktparsers.so.0
#20 0x00007ffff6aff380 in HAC::parser::conditional_instantiation::check_build(HAC::parser::context&) const () from /auto/edatools/hackt/20160622/lib/hackt/libhacktparsers.so.0
#21 0x00007ffff6af0991 in HAC::parser::node_list<HAC::parser::def_body_item const>::check_build(HAC::parser::context&) const () from /auto/edatools/hackt/20160622/lib/hackt/libhacktparsers.so.0
#22 0x00007ffff6aedd07 in HAC::parser::process_def::check_build(HAC::parser::context&) const () from /auto/edatools/hackt/20160622/lib/hackt/libhacktparsers.so.0
#23 0x00007ffff6b773b1 in HAC::parser::node_list<HAC::parser::root_item const>::check_build(HAC::parser::context&) const () from /auto/edatools/hackt/20160622/lib/hackt/libhacktparsers.so.0
#24 0x00007ffff6b73989 in HAC::parser::root_body::check_build(HAC::parser::context&) const () from /auto/edatools/hackt/20160622/lib/hackt/libhacktparsers.so.0
#25 0x00007ffff6b79344 in HAC::parser::imported_root::check_build(HAC::parser::context&) const () from /auto/edatools/hackt/20160622/lib/hackt/libhacktparsers.so.0
#26 0x00007ffff6b7a9ec in HAC::parser::node_list<HAC::parser::imported_root const>::check_build(HAC::parser::context&) const () from /auto/edatools/hackt/20160622/lib/hackt/libhacktparsers.so.0
#27 0x00007ffff6b798d6 in HAC::parser::imported_root_list::check_build(HAC::parser::context&) const () from /auto/edatools/hackt/20160622/lib/hackt/libhacktparsers.so.0
#28 0x00007ffff6b773b1 in HAC::parser::node_list<HAC::parser::root_item const>::check_build(HAC::parser::context&) const () from /auto/edatools/hackt/20160622/lib/hackt/libhacktparsers.so.0
[... cut ...]
#36 0x00007ffff7bc83a6 in HAC::parse_and_check(char const*, HAC::compile_options const&) () from /auto/edatools/hackt/20160622/lib/hackt/libhacktlib.so.0
#37 0x00007ffff7bce3eb in HAC::compile::make_module(int, char**, HAC::compile_options&, util::memory::count_ptr<HAC::entity::module, util::memory::delete_tag>&) () from /auto/edatools/hackt/20160622/lib/hackt/libhacktlib.so.0
#38 0x00007ffff7bced75 in HAC::compile::main(int, char**, HAC::global_options const&) () from /auto/edatools/hackt/20160622/lib/hackt/libhacktlib.so.0
#39 0x00000000004007c2 in main ()

fangism, 2016-06-30:
```
Version: hackt 0.1.4-devel-20160622
CVS Tag: HACKT-00-01-04-main-00-86-59
Configured with: '--with-readline' '--disable-static' 'CC=clang-3.5' 'CXX=clang++-3.5' 'CXXFLAGS=-Wno-error=sign-compare -Wno-error=sign-conversion -Wno-error=unused-function -Wno-error=conversion -Qunused-arguments' 'YACC=/usr/local/experimental/bin/yacc' 'LEX=/usr/local/experimental/bin/flex' '--prefix=/auto/edatools/hackt/20160622' '-C' --enable-ltdl-install
build-triplet: x86_64-unknown-linux-gnu
c++: Ubuntu clang version 3.5.0-4ubuntu2~trusty2 (tags/RELEASE_350/final) (based on LLVM 3.5.0)
AM_CPPFLAGS: -I../../../src     
AM_CXXFLAGS: -pipe -ansi -pedantic-errors -Wold-style-cast -Woverloaded-virtual -W -Wextra -Wall -Wundef -Wshadow -Wno-unused-parameter -Wpointer-arith -Wcast-qual -Wcast-align -Wconversion -Werror
AM_LDFLAGS: 
config-CXXFLAGS: -Wno-error=sign-compare -Wno-error=sign-conversion -Wno-error=unused-function -Wno-error=conversion -Qunused-arguments
config-CPPFLAGS: 
config-LDFLAGS: 
config-LIBS: -ldl 
lex: /usr/local/experimental/bin/flex version 2.5.4
yacc: bison (GNU Bison) 2.7.12-4996
readline: GNU readline 6.3
guile: none
build-date: Wed Jun 22 16:07:58 PDT 2016
```