fangism/hackt #59: Assertion failure

nbingham1, 2016-07-12:
```
export template <pint M, N>
defchan eMx1ofN <: chan(int) () {
}

template <pint N>
typedef eMx1ofN<N, 1> e1of;

export template <pint N>
defproc proc(e1of<N> L) {
}

e1of<2> L;
proc<2> raw(L);
```

```
haco: ../../../src/Object/type/template_actuals.cc:296: HAC::entity::template_actuals
HAC::entity::template_actuals::transform_template_actuals(const this_type &, const
HAC::entity::template_formals_manager &) const: Assertion `a.is_constant()' failed.
Aborted (core dumped)
```

nbingham1, 2016-07-12:
```
Starting program: /auto/edatools/hackt/20160711/bin/haco test.prs
haco: ../../../src/Object/type/template_actuals.cc:296: HAC::entity::template_actuals HAC::entity::template_actuals::transform_template_actuals(const this_type &, const HAC::entity::template_formals_manager &) const: Assertion `a.is_constant()' failed.

Program received signal SIGABRT, Aborted.
0x00007ffff40d2c37 in __GI_raise (sig=sig@entry=6) at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
56	../nptl/sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  0x00007ffff40d2c37 in __GI_raise (sig=sig@entry=6) at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
#1  0x00007ffff40d6028 in __GI_abort () at abort.c:89
#2  0x00007ffff40cbbf6 in __assert_fail_base (fmt=0x7ffff421c3b8 "%s%s%s:%u: %s%sAssertion `%s' failed.\n%n", assertion=assertion@entry=0x7ffff738961f "a.is_constant()", 
    file=file@entry=0x7ffff73d753c "../../../src/Object/type/template_actuals.cc", line=line@entry=296, 
    function=function@entry=0x7ffff73d7758 "HAC::entity::template_actuals HAC::entity::template_actuals::transform_template_actuals(const this_type &, const HAC::entity::template_formals_manager &) const")
    at assert.c:92
#3  0x00007ffff40cbca2 in __GI___assert_fail (assertion=0x7ffff738961f "a.is_constant()", file=0x7ffff73d753c "../../../src/Object/type/template_actuals.cc", line=296, 
    function=0x7ffff73d7758 "HAC::entity::template_actuals HAC::entity::template_actuals::transform_template_actuals(const this_type &, const HAC::entity::template_formals_manager &) const") at assert.c:101
#4  0x00007ffff70d2f00 in HAC::entity::template_actuals::transform_template_actuals(HAC::entity::template_actuals const&, HAC::entity::template_formals_manager const&) const ()
   from /auto/edatools/hackt/20160711/lib/hackt/libhacktparsers.so.0
#5  0x00007ffff6d10215 in HAC::entity::process_definition_alias::make_canonical_fundamental_type_reference(HAC::entity::template_actuals const&) const ()
   from /auto/edatools/hackt/20160711/lib/hackt/libhacktparsers.so.0
#6  0x00007ffff70e1504 in HAC::entity::process_type_reference::make_canonical_process_type_reference() const () from /auto/edatools/hackt/20160711/lib/hackt/libhacktparsers.so.0
#7  0x00007ffff70e154d in HAC::entity::process_type_reference::canonical_compare_result_type::canonical_compare_result_type(HAC::entity::process_type_reference const&, HAC::entity::process_type_reference const&) () from /auto/edatools/hackt/20160711/lib/hackt/libhacktparsers.so.0
#8  0x00007ffff70e1a78 in HAC::entity::process_type_reference::may_be_connectibly_type_equivalent(HAC::entity::fundamental_type_reference const&) const ()
   from /auto/edatools/hackt/20160711/lib/hackt/libhacktparsers.so.0
#9  0x00007ffff71a9f8f in HAC::entity::meta_instance_reference<HAC::entity::process_tag>::may_be_type_equivalent(HAC::entity::meta_instance_reference_base const&) const ()
   from /auto/edatools/hackt/20160711/lib/hackt/libhacktparsers.so.0
#10 0x00007ffff70c0105 in HAC::entity::port_formals_manager::certify_port_actuals(std::vector<util::memory::count_ptr<HAC::entity::meta_instance_reference_base, util::memory::delete_tag>, std::allocator<util::memory::count_ptr<HAC::entity::meta_instance_reference_base, util::memory::delete_tag> > > const&) const () from /auto/edatools/hackt/20160711/lib/hackt/libhacktparsers.so.0
#11 0x00007ffff6d0c776 in HAC::entity::process_definition::certify_port_actuals(std::vector<util::memory::count_ptr<HAC::entity::meta_instance_reference_base, util::memory::delete_tag>, std::allocator<util::memory::count_ptr<HAC::entity::meta_instance_reference_base, util::memory::delete_tag> > > const&) const () from /auto/edatools/hackt/20160711/lib/hackt/libhacktparsers.so.0
#12 0x00007ffff6d0c7b1 in virtual thunk to HAC::entity::process_definition::certify_port_actuals(std::vector<util::memory::count_ptr<HAC::entity::meta_instance_reference_base, util::memory::delete_tag>, std::allocator<util::memory::count_ptr<HAC::entity::meta_instance_reference_base, util::memory::delete_tag> > > const&) const () from /auto/edatools/hackt/20160711/lib/hackt/libhacktparsers.so.0
#13 0x00007ffff6af9741 in HAC::parser::connection_statement::make_port_connection(std::vector<HAC::entity::meta_reference_union, std::allocator<HAC::entity::meta_reference_union> > const&, util::memory::count_ptr<HAC::entity::meta_instance_reference_base const, util::memory::delete_tag> const&) () from /auto/edatools/hackt/20160711/lib/hackt/libhacktparsers.so.0
#14 0x00007ffff6af8df8 in HAC::parser::actuals_base::add_instance_port_connections(util::memory::count_ptr<HAC::entity::meta_instance_reference_base const, util::memory::delete_tag> const&, HAC::parser::context&) const () from /auto/edatools/hackt/20160711/lib/hackt/libhacktparsers.so.0
#15 0x00007ffff6afc519 in HAC::parser::instance_connection::check_build(HAC::parser::context&) const () from /auto/edatools/hackt/20160711/lib/hackt/libhacktparsers.so.0
#16 0x00007ffff6b06a51 in HAC::parser::node_list<HAC::parser::instance_base const>::check_build(HAC::parser::context&) const () from /auto/edatools/hackt/20160711/lib/hackt/libhacktparsers.so.0
#17 0x00007ffff6afbf6b in HAC::parser::instance_declaration::check_build(HAC::parser::context&) const () from /auto/edatools/hackt/20160711/lib/hackt/libhacktparsers.so.0
[... cut ...]
#21 0x00007ffff7bc83a6 in HAC::parse_and_check(char const*, HAC::compile_options const&) () from /auto/edatools/hackt/20160711/lib/hackt/libhacktlib.so.0
#22 0x00007ffff7bce3eb in HAC::compile::make_module(int, char**, HAC::compile_options&, util::memory::count_ptr<HAC::entity::module, util::memory::delete_tag>&) ()
   from /auto/edatools/hackt/20160711/lib/hackt/libhacktlib.so.0
#23 0x00007ffff7bced75 in HAC::compile::main(int, char**, HAC::global_options const&) () from /auto/edatools/hackt/20160711/lib/hackt/libhacktlib.so.0
#24 0x00000000004007f2 in main ()
```

fangism, 2016-07-12:
Similar test case:

```
template <pint N>
defchan eMx1ofN <: chan(int) () { }

template <pint N>
typedef eMx1ofN<N> e1of;

template <pint N>
defproc proc(e1of<N> L) { }

e1of<2> L;
proc<2> raw;
raw.L = L;
```

fangism, 2016-07-12:
Also:

```
template <pint N>
defchan eMx1ofN <: chan(int) () { }

template <pint N>
typedef eMx1ofN<N> e1of;

template <pint N>
defproc proc(e1of<N> L) { }

proc<2> x;
x.L = y.L;
```

fangism, 2016-07-13:
sample trace:

```
type l: bar<proc::N>
type r: bar<proc::N>
\-{ virtual count_ptr<const HAC::entity::process_type_reference> HAC::entity::process_definition_alias::make_canonical_fundamental_type_reference(const HAC::entity::template_actuals &) const
| \-{ HAC::entity::template_actuals HAC::entity::template_actuals::transform_template_actuals(const this_type &, const HAC::entity::template_formals_manager &) const
<proc::N>
lt-haco: ../../../src/Object/type/template_actuals.cc:299: HAC::entity::template_actuals HAC::entity::template_actuals::transform_template_actuals(const this_type &, const HAC::entity::template_formals_manager &) const: Assertion `a.is_constant()' failed.
```

fangism, 2016-07-13:
Findings:
This is executing code in a path that really needs massive rewriting.
The template typedef forwarding mechanism is faulty in that it is expecting resolved constants at pre-elaboration compile time when elaborated values are not guaranteed to be available.
The code in the trace is attempting to check type equivalence (may-analysis) too early, and I don't have good error handling in place to allow resolution to fail.
The best we could do now is alter the may-equivalence check to always (conservatively) return true, but that would be too loose at this time, missing obvious cases where the base definition is not matched.  Today, if we punt all checks to elaboration (create) time, we unfortunately lose source:line number information.